chenxuuu/llcom #196: 不可见字符是否能优化一下，现在有点看不太清楚

zhaozhaofox, 2025-04-16:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

不可见字符是否能优化一下，现在有点看不太清楚

### 描述您想要的解决方案 / Describe the solution you'd like

不可见字符是否能优化一下，现在有点看不太清楚

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

不可见字符是否能优化一下，现在有点看不太清楚

LuaRobot, 2025-04-16:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供功能需求的详细描述 does not provide detailed description of the feature request

chenxuuu, 2025-04-16:
有什么建议的方案吗？

zhaozhaofox, 2025-04-16:
可以参考：标准 ASCII 码对照表的缩写栏位，\x00 -> NUL, \x03 -> ETX。当然控制格式的\r\n \t之类的不用这样

[image]

**就像这样**
[image]

chenxuuu, 2025-04-16:
这只是不同字体的显示效果，并不是作了特殊处理

[image]

chenxuuu, 2025-04-16:
你要是可以提供一个字体的话，可以尝试嵌入进去看看效果