chenxuuu/llcom #198: release to scoop for windows?

konosubakonoakua, 2025-05-19:
### 您的功能请求是否与解决某些问题有关？请描述一下。/ Is your feature request related to a problem? Please describe.

as the title

### 描述您想要的解决方案 / Describe the solution you'd like

contribute to scoop buckets

### 描述您想要的详细使用步骤描述 / Describe the solution you'd like to use in what way

scoop install llcom

LuaRobot, 2025-05-19:
⚠ 你的Issue可能缺少了一些信息，请自觉补全，以证明issue内容的真实性：

⚠ Your issue may be missing some information, please complete it to prove the authenticity of the content of the issue:

- ❌未提供功能需求的详细描述 does not provide detailed description of the feature request

chenxuuu, 2025-05-19:
等llcom2写完之后再加吧，不然感觉会乱

chenxuuu, 2026-11-06:
@LuaRobot 你研究下这个的可行性和可能的坑

LuaRobot, 2026-11-06:
按 @chenxuuu 的说法把可行性摸了一遍：**形态很干净，技术上可行，但现在卡在发布渠道上**，另外有几个真坑是代码层面的，得先定一下怎么处理。

## 先说好的部分

- release 包是纯便携 zip，解压即用：官网 `llcom.zip`（4.41 MB）里只有 `llcom.exe` + `core_script/` 五个 lua，没有安装器、没有注册表、没有服务。GitHub 那份 `llcom_x64.zip` 更极端，只有一个 exe（Costura 打包）。
- 不需要管理员：代码里没有 `requireAdministrator`，配置直接写自身目录（`Global.cs:161` 普通 exe 时 `ProfilePath = AppPath`），装在 Scoop 的用户目录下正好。
- 只有 x64（`serial_monitor.dll` / `xlua.dll` 分了 costura32/64，release 只发 x64），标 `64bit` 就行。
- License 是 Apache-2.0，`.NET Framework 4.6.2` 是 Win10 1809 起自带的，不用写 `depends`。

## 卡住的地方：没有版本化的下载地址

- GitHub release 停在 **1.1.3.9**（2025-01-25），而 tag 已经有 `1.1.4.0/1.1.4.1/1.1.4.2`，都没有发 release 资产。
- 官网 `https://llcom.papapoi.com/llcom.zip` 是 **1.1.4.2**（我下了，exe 版本 1.1.4.2，sha256 `b17752a7…d8ec1d`），但 URL 里没有版本号。

Scoop 的硬要求是「一个版本对应一个不会变的 URL + 可复核的 hash」。同 URL 原地换内容，装过的人 `scoop install` 就会 hash 不匹配，excavator 也没法自动升。所以要么把 1.1.4.x 发到 GitHub release（推荐，URL 天然版本化），要么官网提供 `llcom-1.1.4.2.zip` 这种带版本的文件名。

checkver 本身好办，指向现成的 `https://llcom.papapoi.com/autoUpdate.xml` 用 `xpath: /item/version` 就行，autoupdate 的 hash 也能用 `{"url": "$url", "mode": "download"}` 现算。

## 坑（按严重程度）

1. **内置自动更新会跟 Scoop 打架**。`setting.autoUpdate` 默认 `true`（`Model/Settings.cs:36`），启动就查 `autoUpdate.xml` 并弹 `ShowUpdateForm`，确认后直接覆盖程序目录的 exe → Scoop 记的版本和实际 exe 对不上。解法是在 manifest 里 `post_install` 写一份 `{"autoUpdate": false}` 的 settings.json（其余字段走 Model 默认值），并在 `notes` 里说明用 `scoop update`。

2. **Scoop 的 `bin` shim 会触发「不支持同文件夹多开」**。多开检测是 `GetProcessesByName(ProcessName)` 数量 > 1 且 `lock` 存在就 `Environment.Exit(1)`（`Global.cs:254-260`），而 Scoop 的 shim 文件名同样是 `llcom.exe`、且会 `WaitForSingleObject(INFINITE)` 等子进程退出（`ScoopInstaller/Shim` 的 `cpp/shim.cpp:988`）。所以从第二次通过 shim 启动开始，进程表里必然 ≥2 个 `llcom.exe`，会直接报错退出。→ 短期解法是 manifest 只写 `shortcuts`（直接指 exe），不写 `bin`；长期要改代码的话，把检测换成按 `MainModule.FileName` 比目录、或排除自身 PID。

3. **persist 不能无脑加**。配置全在 exe 同目录，Scoop 升级会换目录，所以 `settings.json`、`user_script_run`、`user_script_send_convert`、`user_script_recv_convert`、`logs` 必须 persist。但 `settings.json` 若被建成空文件，llcom 解析失败会弹「配置文件加载失败」然后 `Environment.Exit(1)`（`Global.cs:174-181`）→ persist 时必须 `pre_install` 里先写一个 `{}`。`core_script/` 建议**不** persist：`head.lua`/`sys.lua` 每次启动强制覆盖，其余「存在就不写」，随包发新版核心脚本才生效。

4. 两个小雷：exe 文件名必须还是 `llcom.exe`（`Global.cs:227`，改了就退出），路径里不能含 `\AppData\Local\Temp\` 或 `\Windows\Temp\`（`Global.cs:232`）——Scoop 默认目录都没问题，但用户把 scoop 目录配到 Temp 下会直接起不来。

## 草稿（还没跑过，本机是 Linux 没 Scoop）

```json
{
    "version": "1.1.4.2",
    "description": "串口调试助手，支持 Lua 脚本扩展",
    "homepage": "https://llcom.papapoi.com/",
    "license": "Apache-2.0",
    "architecture": {
        "64bit": {
            "url": "https://llcom.papapoi.com/llcom-1.1.4.2.zip",
            "hash": "sha256:b17752a78e0d685119ea172d0c7b100577d7125a98f4fd25e65bff5d73d8ec1d"
        }
    },
    "shortcuts": [["llcom.exe", "llcom"]],
    "persist": ["settings.json", "logs", "user_script_run", "user_script_send_convert", "user_script_recv_convert"],
    "pre_install": [
        "if (!(Test-Path \"$persist_dir\")) { New-Item -ItemType Directory -Path \"$persist_dir\" | Out-Null }",
        "if (!(Test-Path \"$persist_dir\\settings.json\")) { Set-Content -Path \"$persist_dir\\settings.json\" -Value '{}' -Encoding UTF8 }"
    ],
    "post_install": "'{\"autoUpdate\":false}' | Set-Content -Path \"$dir\\settings.json\" -Encoding UTF8",
    "checkver": { "url": "https://llcom.papapoi.com/autoUpdate.xml", "xpath": "/item/version" },
    "autoupdate": {
        "architecture": { "64bit": { "url": "https://llcom.papapoi.com/llcom-$version.zip", "hash": { "url": "$url", "mode": "download" } } }
    },
    "notes": "需要 .NET Framework 4.6.2+（Win10 1809 起自带）。Scoop 版已关掉内置自动更新，升级请用 scoop update llcom。"
}
```

（`url` 里的 `llcom-$version.zip` 是假设官网会给版本化文件名；如果改走 GitHub release，把 URL 换成 release 资产、`checkver` 换成 `github` 更省事。）

下一步等你定：**发布渠道走 GitHub release 还是官网版本化文件名？** 定了我就能把 manifest 补完、跑 `checkver`/hash 校验，再往 Extras 提 PR（GUI 工具在 Extras，不在 Main）。另外如果同意改两行代码（多开检测按目录判、加个 `--no-auto-update` 或环境变量），坑 1、2 都能从根上消掉，比在 manifest 里绕干净。